bruvv/LILYGO-T-Higrow-Esphome #19: Add DHT12 sensor support

P1X3L8, 2023-09-11:
All my T-Higrow boards use a DHT12 instead of the DNT11. After calibration, the values returned vary considerably and are unreliable for monitoring. Not sure if the DHT12 uses I2C instead of GPIO.

bruvv, 2023-09-12:
I fixed this issue I think, can you test the new firmware?

pando85, 2024-03-10:
For me it doesn't work. It detects it as DHT11.

DHT12 seems a different platform: https://esphome.io/components/sensor/dht12.html but it doesn't work.

P1X3L8, 2024-03-10:
Same for me, detects as DHT11

bruvv, 2024-06-04:
Sorry for the late reply, I added dht12: https://github.com/bruvv/LILYGO-T-Higrow-Esphome/blob/main/LILYGO-T-Higrow-ESP32.yaml#L113 but I do not that sensor so can any of you guys test it?

P1X3L8, 2024-06-04:
bruvv no worries. Your changes won't work. I tried fixing this myself some time ago and found that the esphome DHT12 platform doesn't support 1-wire mode. You have to use DHT11 platform with the DHT12 sensor. This isn't ideal and the accuracy/reliability is poor. I've given up on these boards for now and am looking for another solution.

-PiXEL8

closed 2024-06-04 by bruvv