bruvv/LILYGO-T-Higrow-Esphome #23: How to make changes in yaml config

MarnixMeersman, 2024-06-04:
Hi, 
I hope this is the right platform. 
I have a lilygo with a BME280. I assumed that forking this repo, then making adjustments and flashing the esp with my forked version would work. 

However, it keeps searching for a DHT sensor instead of BME. 
[image]
[image]

bruvv, 2024-06-04:
Can you show me what you changed?

MarnixMeersman, 2024-06-04:
ofcourse, 
This is my forked YAML config:
```
---
substitutions:
  # Dashes are not supported in devicename
  devicename: 'lilygohigrowplantsensor'
  device_description: 'Keep my plants safe and lively'
  project_version: '1.1'
  update_interval: 30min
  loglevel: DEBUG
  moisture_min: '2.82'
  moisture_max: '1.39'
  conductivity_min: '0.075'
  conductivity_max: '0.25'
  # Uncomment run_duration and sleep_duration if you want to use deepsleep
  # set how long to stay awake - NOT less then 10sec
  # run_duration: 11s
  # set how long to sleep in minutes
  # sleep_duration: 60min

esphome:
  name: '${devicename}'
  comment: '${device_description}'
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: true

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: esphome.project-template
    version: '${project_version}'
  on_boot:
    priority: 240
    then:
      - wait_until:
          condition:
            wifi.connected:
          timeout: 10s
  on_shutdown:
    then:
      - switch.turn_off: spower

dashboard_import:
  package_import_url: github://bruvv/LILYGO-T-Higrow-Esphome/LILYGO-T-Higrow-ESP32.yaml@main

esp32:
  board: lolin_d32

improv_serial:

wifi:
  # Use improv (after installing either go to web.esphome.io or use the installer here: https://bruvv.github.io/LILYGO-T-Higrow-Esphome)
  # ssid: !secret wifi_ssid
  # password: !secret wifi_password
  # fast_connect: True

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${devicename}'

captive_portal:

# Web server disabled because it was using to much ram
# web_server:
# port: 80

# Enable logging
logger:
  level: '${loglevel}'

api:

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

time:
  - platform: homeassistant

button:
  - platform: restart
    name: 'Restart - ${devicename}'

i2c:
  sda: 25
  scl: 26
  scan: true
  id: bus_a
  setup_priority: -200

switch:
  # Power Switch
  - platform: gpio
    name: '${devicename} Sensor Power switch'
    pin:
      number: 4
      mode: INPUT_PULLUP
    id: spower
    restore_mode: ALWAYS_ON
    internal: true
    setup_priority: 1000

sensor:
  # Wifi sensor
  - platform: wifi_signal
    name: '${devicename} WiFi Signal'
    id: '${devicename}_wifi_signal'
    update_interval: ${update_interval}

packages:
  text_sensors: !include common/text_sensors.yaml
  # dht: !include common/dht.yaml
  plantsensors: !include common/plantsensors.yaml
  # waterpump: !include common/waterpump.yaml
  bluetooth: !include common/bluetooth.yaml
  # Battery only works for 12 hours with deepsleep!
  bme280: !include common/bme280.yaml
  # deepsleep: !include common/deepsleep.yaml
  battery: !include common/battery.yaml
```
  
  So the only change i the commenting out of the dht config and enebaling the inclusion of BME. 
  
  Here's my forked repo: https://github.com/MarnixMeersman/LILYGO-T-Higrow-Esphome/blob/main/LILYGO-T-Higrow-ESP32.yaml
  
  Thanks!

bruvv, 2024-06-04:
Looks good, how did you flash it? You should clone it, edit it and then run `esphome run LILYGO-T-Higrow-ESP32.yaml` locally

bruvv, 2024-06-04:
It should be fine but yes please connect it with USB and try flashing again

bruvv, 2024-06-08:
ah that I can fix! Please test the new version. Make sure to read the new yaml correctly.

bruvv, 2025-06-02:
i am asuming this is working now. if not please reopen the issue.