bruvv/LILYGO-T-Higrow-Esphome #24: View Both Sensors in battery.yaml: # Batterie volt and # Battery %.

Copyrighter, 2024-06-11:
How can I have both Sensors enabled in battery.yaml: # Batterie volt and # Battery %.?
If I add both of them, I see following error:
`Pin 33 is used in multiple places.`
Each one separately is working.

bruvv, 2024-06-13:
fixed https://github.com/bruvv/LILYGO-T-Higrow-Esphome/commit/5fe36fa56ec550fb31f84db49eec0bc1607d725e

closed 2024-06-13 by bruvv

Copyrighter, 2024-06-13:
@bruvv 
is there any other solution than escaping?
is there option to see both % and V?

bruvv, 2024-06-13:
not that I know of. Mine broke so I cannot test it at the moment :(

Copyrighter, 2024-06-14:
@bruvv 
I found a working fix ( tested ).
change syntax to:
---
  - platform: adc
    pin:
      number: 33
      allow_other_uses: true
...
---

bruvv, 2024-06-14:
nice found! I added it to the repo:)