bruvv/LILYGO-T-Higrow-Esphome #28: Issue with the wakeup pin: in deeplseep.yaml

TobiasMunich, 2024-07-03:
error message is:
in deeplseep.yaml Pin 35 is used in multiple places.

------------------------------------------------------------------------------------------------

deep_sleep: [source /config/esphome/common/deepsleep.yaml:2]
  
  **Pin 35 is used in multiple places.**
  id: deep_sleep_control
  run_duration: 20s
  sleep_duration: 2min
  wakeup_pin: 
    number: 35
    inverted: True
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
binary_sensor.gpio: [source /config/esphome/common/deepsleep.yaml:15]
  
  Pin 35 is used in multiple places.
  platform: gpio
  name: pflanze1 Wake Button
  pin: 
    number: 35
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_pin_validation_error: False

bruvv, 2024-07-03:
Fixed :) please test

TobiasMunich, 2024-07-03:
Tested,
The fix is not working 

Failed config

deep_sleep: [source /config/esphome/common/deepsleep.yaml:1]
  
  Pin 35 is used in multiple places.
  id: deep_sleep_control
  run_duration: 60s
  sleep_duration: 30min
  wakeup_pin: 
    number: 35
    inverted: True
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False

bruvv, 2024-07-03:
Try again please, i just pushed another update.

TobiasMunich, 2024-07-03:
this was fast ;-)
perfect it works!

closed 2025-06-02 by bruvv